sot/mica #65: aca_dark update broken for skare 0.18

jeanconn, 2015-06-09:
```
File "/proj/sot/ska/arch/x86_64-linux_CentOS-6/lib/python2.7/site-packages/mica/archive/aca_dark/update_aca_dark.py", line 38, in <module>
    MICA_FILES = pyyaks.context.ContextDict('update_mica_files', basedir='/tmp')
  File "/proj/sot/ska/arch/x86_64-linux_CentOS-6/lib/python2.7/site-packages/pyyaks-3.3.3-py2.7.egg/pyyaks/context.py", line 251, in __new__
    "({} vs. {})".format(name, basedir, CONTEXT[name].basedir))
ValueError: Re-using context name 'update_mica_files' but basedirs don't match (/tmp vs. /data/aca/archive)
```

jeanconn, 2015-06-09:
I see that update_aca_dark.py does

```
from . import dark_cal
```

at

https://github.com/sot/mica/blob/8bdbcc235a4f7615c0ecbd22f5e6c764feb21e47/mica/archive/aca_dark/update_aca_dark.py#L23

And dark_cal.py already seems to define these things for pyyaks

```
MICA_FILES = pyyaks.context.ContextDict('update_mica_files',
                                        basedir=os.path.join(MICA_ARCHIVE_PATH))
MICA_FILES.update(file_defs.MICA_FILES)
```

https://github.com/sot/mica/blob/8bdbcc235a4f7615c0ecbd22f5e6c764feb21e47/mica/archive/aca_dark/update_aca_dark.py#L38

but MICA_FILES are redefined in update_aca_dark.py.  Is there a motivation for the redundancy or can the redefinition in update_aca_dark.py just be stripped out?

taldcroft, 2015-06-10:
Yeah, in `update_aca_dark.py` it should just be:
```
MICA_FILES = dark_cal.MICA_FILES
```
This is exactly the reason I made mismatching basedirs be an exception, because it usually means you are doing something wrong.  I think the `/tmp` dir is red herring, there is a no harm in compiling the script with the basedir pointing at the actual files.

jeanconn, 2015-06-11:
Should any redefinition be an error in pyyaks?  It is a little confusing that the SKA_FILES and DARK_CAL redefinitions aren't errors (I assume because they are no-ops).