sot/mica #73: Should 'downlink' be True or False for a dark cal replica that couldn't be used?

jeanconn, 2015-12-29:
Should 'downlink' be true or false for a replica that couldn't be used?  Does downlink represent 'successful' downlink?  

```
        {
            "ccd_temp": -15.070860927152314,
            "ccd_temp_source": "HDR3",
            "count": 3,
            "downlink": false,
            "start": "2015:355:17:37:09.043",
            "stop": "2015:355:20:38:38.644"
        },
```

taldcroft, 2015-12-29:
`downlink` for replica N just means that there is a file named `'VC2_Replica{}_SFDU'.format(i_replica + 1)`.  It has nothing to do with "could be used" status (except of course a file that doesn't exist can't be used).

In this case there is no replica 3 in the relevant directory.  I don't know that we ever got a full report on why that is the case.

jeanconn, 2015-12-29:
> downlink for replica N just means that there is a file named 'VC2_Replica{}_SFDU'.format(i_replica + 1). It has nothing to do with "could be used" status (except of course a file that doesn't exist can't be used).

I know how the code works to define it, I just don't know how/if the
value is used in anything else.  I didn't see docs on this locally.

> In this case there is no replica 3 in the relevant directory. I don't know that we ever got a full report on why that is the case.

In this case, there is a replica 3, but I renamed it in the data
directory so the IDL processing code would continue (before
implementing https://github.com/sot/aca_dark_cal/commit/9deefae6a4cf458546a455510da7a89b08534419
)

taldcroft, 2015-12-30:
Anyway, feel free to add another key to the mica info if that is useful for you.  The `downlink` flag is just a simple "got some data in the file".  I don't know if anything is using it but that's what it is.

taldcroft, 2015-12-30:
So to close this out, the answer to the original question is:

`downlink` could be True or False for a replica that couldn't be used.

closed 2015-12-30 by taldcroft