sot/mica #77: Code review of "corrected" star positions at acquisition

jeanconn, 2016-02-04:
We should do a review of the code that determines the "corrected" star positions at acquisition to confirm correct/continued use for predicting that a star "would have been acquired" in the context of the multiple star flag.

jeanconn, 2016-02-16:
Question:  Are the attitude quaternions being applied properly to determine expected star positions?

Answer:  Looks good.  JMC was a bit confused about when aca_misalign is applied, but the code is correct (even if it uses an unnecessary identity matrix for aca_misalign) and confusion has been alleviated.  https://github.com/sot/mica/blob/master/mica/stats/acq_stats.py#L121

Question: Is the process to check against proper-motion corrected AGASC position after the one-shot has been applied good enough to use for these post-facto position checks for the "would the star have been identified" idea?  On-board it is using the distance from the tracked star to each other star.

Answer: Yes.

Question: Is the (img_func == 'star') check sufficient in the  "would the star have been identified" code?

Answer: There is a mismatch here between the limit of the corrected position offset (5 arcsec ~ dr) and the on-board id dist limit.  Probably fine.

taldcroft, 2016-02-18:
Looks good!

closed 2016-02-18 by taldcroft