sot/mica #81: Improve multiple aspect interval warnings

jeanconn, 2016-05-10:
For warnings like:

"Warning: obsid 18842 has 2 aspect intervals"

it would be good to included "(has monitor window)" if, in fact, the observation does.  If it doesn't an email warn on this error would be useful.

taldcroft, 2016-05-10:
It seems like there should be no warning if it has a monitor window and 2 aspect intervals, i.e. that is the expected outcome right?  But there would be a warning if it had 1 or > 2.

jeanconn, 2016-05-10:
I don't think we care on warning if the monitor window observation has 1, because hopefully everything else would be broken on such an observation.  But I haven't looked at the code for the implementation details; I'll do whatever is simplest.

And yes, my point was to only draw appropriate attention to warnings that are actually warnings, but I suppose it make more sense to just not warn on them.