sot/mica #153: Add function or option to mica.starcheck to get either Acquisition or Guide catalog

jeanconn, 2017-08-04:
It seems like most of the time I'm fetching the "catalog" from starcheck and then constructing either a guide or acquisition catalog from the rows of the full catalog.  Seems like it might make sense to add this functionality right into the fetching code, either as new functions or as options to `get_starcheck_catalog` and `get_starcheck_catalog_at_date`.

The guide catalog could set the "halfw" to 20 appropriately, avoiding the need to just *know* that implicit value.

The acquisition catalog could add a field to include the index needed to map back to the acquisition id msid.

taldcroft, 2017-08-04:
:+1: for a new option in both.  E.g. `type=(FID|GUI|ACQ|BOT|MON)` that would select catalog entries matching `type`.  And of course a `BOT` star would match `type='ACQ'`.

As an unrelated feature request, it would probably be useful to have a column `size` that is an integer representation of `SZ`.

:+1: also on `halfw=20` for guide always.

jeanconn, 2017-08-04:
I think that "And of course a BOT star would match type='ACQ'" actually gets confusing compared to my idea to have options for the two types of catalogs instead of the 5 types of things.

taldcroft, 2017-08-04:
So you want something like `type='guide_catalog'` that gives you the guide and mon stars and fids, and `type='acq_catalog'` gives all the acquisition stars?

I think there is merit in both (meaning there would be 7 options).  It won't be too confusing as long as the docs are well-written.  :smile:

taldcroft, 2017-08-04:
`type` could just be a regex, where `'guide_catalog'` gets translated to `'BOT|GUI|FID|MON'`.

taldcroft, 2017-08-04:
But of course `type` is a reserved word, so the kwarg would need to be named differently.

jeanconn, 2017-08-04:
:smile:  Yeah, I'll think through this idea before implementing and will document well.

And yes, regarding a regex, something like that could work.  I was just thinking that I can't see a reason for say, type="BOT".  And would not get confused about whether type = "GUI" would go to "GUI" or "GUI|BOT" or "GUI|BOT|FID|MON" , so yes, maybe your 7 options is the most reasonable thing.

jeanconn, 2017-08-04:
However, once I have a set of stars, I don't see a way to generalize, say halfw=20 for the guide phase of a BOT star if I haven't defined which of the two primary types of catalog is actually in play.

jeanconn, 2024-09-26:
Still a good idea but if we haven't needed it by now...

closed 2024-09-26 by jeanconn