sot/mica #169: Include star `ra_pmcorr` and `dec_pmcorr` in guide stats table

taldcroft, 2018-01-30:
The guide stats table would be more useful if it had the PM-corrected RA and dec.  Those are pretty *always* what you want from the table.  It is a little risky even having the `ra` and `dec` columns, since it is very easy to just use those without realizing they are wrong.

jeanconn, 2018-01-30:
Sure.  Sounds good.  

I think I've never used star positions from this table so hadn't noticed this defect (if working from a catalog, I have yag zag and quaternion and if working with individual stars from rows of the table I have agasc_ids).

jeanconn, 2018-01-30:
Oops, I think this might be a documentation problem instead of a defect:

https://github.com/sot/mica/blob/745efc434aae1a5df29f83697f91e9e32149fc97/mica/stats/guide_stats.py#L473

taldcroft, 2018-01-30:
Ah, perfect!  I admit I didn't look at the docs but figured all those AGASC cols would be just copied straight through.

BTW, catalog yag/zag is specified only to the nearest arcsec, so that is commonly not good enough.

closed 2018-01-30 by taldcroft

jeanconn, 2018-01-31:
Good point about catalog yag/zag.  In this code of course it is using deltas from the star positions determined from their pm-corrected RA/Dec and the OBC quaternions.  And at some point that should be updated to use the CentroidResiduals code.

Hmm.  I wonder if it would be more useful to have the calculated yang/zang at for the commanded quaternion (to more places) in the table instead of the catalog values... but probably overkill.

taldcroft, 2018-01-31:
> I wonder if it would be more useful to have the calculated yang/zang at for the commanded quaternion (to more places) in the table instead of the catalog values... but probably overkill.

That would definitely be useful, at least in my latest use case I have to compute that on the side.