sot/mica #179: Test failure running as not jeanconn

taldcroft, 2018-07-26:
Presumably I could have the right credentials, but this should probably just be skipped if the credentials file isn't there.
```
=================================== FAILURES ===================================
______________________________ test_write_reports ______________________________

    @pytest.mark.skipif('not HAS_SYBASE_ACCESS', reason='Report test requires Sybase/OCAT access')
    @pytest.mark.skipif('not HAS_SC_ARCHIVE', reason='Report test requires mica starcheck archive')
    def test_write_reports():
        """
        Make a report and database
        """
        tempdir = tempfile.mkdtemp()
        # Get a temporary file, but then delete it, because report.py will only
        # make a new table if the supplied file doesn't exist
        fh, fn = tempfile.mkstemp(dir=tempdir, suffix='.db3')
        os.unlink(fn)
        report.REPORT_ROOT = tempdir
        report.REPORT_SERVER = fn
        for obsid in [20001, 15175, 54778]:
>           report.main(obsid)

../../../../x86_64-linux_CentOS-6/lib/python2.7/site-packages/mica/report/tests/test_write_report.py:35: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../../x86_64-linux_CentOS-6/lib/python2.7/site-packages/mica/report/report.py:501: in main
    links = obs_links(obsid, summary['seq_nbr'], summary['lts_lt_plan'])
../../../../x86_64-linux_CentOS-6/lib/python2.7/site-packages/mica/report/report.py:342: in obs_links
    vv, vvid = official_vv(obsid)
../../../../x86_64-linux_CentOS-6/lib/python2.7/site-packages/mica/report/report.py:262: in official_vv
    vv_db = Ska.DBI.DBI(dbi='sybase', server='sqlsao', user='jeanconn', database='axafvv')
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <Ska.DBI.DBI.DBI object at 0x7ff9138e8d10>, dbi = 'sybase'
server = 'sqlsao', user = 'jeanconn', passwd = None, database = 'axafvv'
numpy = True, autocommit = True, verbose = False
authdir = '/proj/sot/ska/data/aspect_authorization', kwargs = {}
DEFAULTS = {'sqlite': {'server': 'db.sql3'}, 'sybase': {'database': 'aca', 'server': 'sybase', 'user': 'aca_ops'}}
dbapi2 = <module 'Sybase' from '/proj/sot/ska/arch/x86_64-linux_CentOS-6/lib/python2.7/site-packages/python_sybase-0.39-py2.7-linux-x86_64.egg/Sybase.pyc'>
passwd_file = '/proj/sot/ska/data/aspect_authorization/sqlsao-axafvv-jeanconn'
e = IOError(13, 'Permission denied')
```

jeanconn, 2018-07-26:
Sure.  Though I thought last time we were here (and I may be wrong) we'd decided that if you are running it on HEAD and should have access to everything, not running all the tests is a fail (so you should just have the access file).

jeanconn, 2018-07-26:
Though given my name is all over that, I'm guessing I've "jeanconn" just coded in report.py (which would be an even bigger error).

taldcroft, 2018-07-26:
> Sure. Though I thought last time we were here (and I may be wrong) we'd decided that if you are running it on HEAD and should have access to everything

Agreed.  I don't know off the top of my head what those credentials are and if I have them (you can tell me offline), but still the mica tests should pass/skip for non-privileged users.