sot/mica #185: Invalid path returned in Windows (for network paths)

chandra-xray, 2019-01-17:
In Windows, when a user evaluates this method from a network location (e.g. \\\\Noodle\\Greta\\mission) the path returned by the line:

https://github.com/sot/mica/blob/adf637287963285b2f078294c4af578112ed239a/mica/archive/aca_dark/dark_cal.py#L54

is invalid/incorrect.  It will either return, Z:\ (which is not actually mapped on the given machine) or \\\\Noodle\\Greta\\ (which is the current directory, but not the directory that should be used for this.

I believe this issue is actually with the pyyaks package not using the basedir specified for the absolute path, but I haven't been able to trace the exact issue.

jeanconn, 2019-01-17:
Not sure if pyyaks here

https://github.com/sot/pyyaks/blob/ea6dc810f0634f4e106b8c8377b722601d8bcd08/pyyaks/context.py#L197

might need the same treatment as https://github.com/sot/mica/commit/ae00c60510e28c0852ac6ad14bab86563b5e1399 or if that would be unrelated.

taldcroft, 2019-01-18:
Can you try using the fix in https://github.com/sot/pyyaks/pull/6?

BTW who made the issue?  Is that Mark B?

jeanconn, 2019-01-18:
I assume @mbaski would have posted with his credentials, and since the chandra-xray user has the aca email address I assume the posting user is unlikely to see notifications.

chandra-xray, 2019-01-18:
@jskrist made this issue.

taldcroft, 2019-01-18:
Great.  @jskrist can you try testing with that pyyaks fix?

jskrist, 2019-01-18:
Sorry for the confusion. I logged in as @chandra-xray  to make sure I had access to all the source code repos while I was investigating the issue, then made the issue using the same account.

jskrist, 2019-01-18:
@taldcroft , I’m out today, but I can test it later this afternoon or Monday morning if it’s not urgent.

taldcroft, 2019-01-18:
@jskrist - no rush.

jskrist, 2019-01-22:
To close the loop:  [I reviewed the fix](https://github.com/sot/pyyaks/pull/6#pullrequestreview-194661510) in sot/pyyaks#6

closed 2019-01-28 by jeanconn