sot/mica #195: mica.starcheck for 20919 is wrong

jeanconn, 2019-06-11:
20919 does not have ACQ only stars
```
sqlite> select * from starcheck_catalog where obsid = 20919;
2214|20919|1|2019:147:00:45:56.704|1|0|1||FID|8x8|||7.0|8.0|919|-844|1|1|25|--|--
2214|20919|1|2019:147:00:45:56.704|2|1|5||FID|8x8|||7.0|8.0|-1828|1053|1|1|25|--|--
2214|20919|1|2019:147:00:45:56.704|3|2|6||FID|8x8|||7.0|8.0|385|1697|1|1|25|--|--
2214|20919|1|2019:147:00:45:56.704|4|3|548932512||BOT|6x6|0.977||8.57|10.078|-727|1492|28|1|160|--|--
2214|20919|1|2019:147:00:45:56.704|5|4|548933960||BOT|6x6|0.978||8.497|10.0|1757|-635|28|1|160|--|--
2214|20919|1|2019:147:00:45:56.704|6|5|548937184||BOT|6x6|0.889||9.89|11.391|101|-652|20|1|120|--|--
2214|20919|1|2019:147:00:45:56.704|7|6|548939872||BOT|6x6|0.722||10.266|11.766|-575|-2003|8|1|60|--|--
```

If I regenerate the full database with no code changes,it looks fine
```sqlite> select * from starcheck_catalog where obsid = 20919;
2214|20919|1|2019:147:00:45:56.704|1|0|1||FID|8x8|||7.0|8.0|919|-844|1|1|25|--|--
2214|20919|1|2019:147:00:45:56.704|2|1|5||FID|8x8|||7.0|8.0|-1828|1053|1|1|25|--|--
2214|20919|1|2019:147:00:45:56.704|3|2|6||FID|8x8|||7.0|8.0|385|1697|1|1|25|--|--
2214|20919|1|2019:147:00:45:56.704|4|3|548932512||BOT|6x6|0.977||8.57|10.078|-727|1492|28|1|160|--|--
2214|20919|1|2019:147:00:45:56.704|5|4|548933960||BOT|6x6|0.978||8.497|10.0|1757|-635|28|1|160|--|--
2214|20919|1|2019:147:00:45:56.704|6|5|548937184||BOT|6x6|0.889||9.89|11.391|101|-652|20|1|120|--|--
2214|20919|1|2019:147:00:45:56.704|7|6|548939872||BOT|6x6|0.722||10.266|11.766|-575|-2003|8|1|60|--|--
2214|20919|1|2019:147:00:45:56.704|8|7|549849072||BOT|6x6|0.886||9.39|10.891|-2390|-847|28|1|160|--|--
2214|20919|1|2019:147:00:45:56.704|9|0|548936840||ACQ|6x6|0.699||10.292|11.797|1637|-607|8|1|60|--|--
2214|20919|1|2019:147:00:45:56.704|10|1|548935824||ACQ|6x6|0.425||10.555|12.062|-767|2309|8|1|60|--|--
2214|20919|1|2019:147:00:45:56.704|11|2|548938256||ACQ|6x6|0.399||10.58|12.078|-693|-2244|8|1|60|--|--
```

I don't know if there might be a race condition or some such in database update.  @taldcroft  what steps would we need to take to validate / promote a starcheck.db3 replacement... (such as me just copying the file I just made into the mica archive)?

taldcroft, 2019-06-11:
Copying the new file is fine since you did not modify any code.

jeanconn, 2019-06-11:
I wasn't sure if I needed to compare all observations or do any other validation.

jeanconn, 2019-06-11:
I'll backup current database and copy the new one in.

closed 2019-06-11 by jeanconn