sot/mica #204: starcheck_parser is now in starcheck

jeanconn, 2019-08-05:
Since starcheck parser got moved into starcheck, it looks like proseco's get_kwargs_from_starcheck_text and some annie test pieces that want mica.starcheck.starcheck_parser are going to be broken on systems without starcheck.    

That's not a problem in envs with starcheck/perl/production envs, but might be undesireable overall.

taldcroft, 2019-08-05:
Can we just add the `starcheck` package to `ska3-flight` and `ska3-matlab`?

jeanconn, 2019-08-05:
Do you mean build the starcheck conda package without the dependencies
called out (like Perl that isn't built on 32 bit)?  I'm not thrilled with
that.

taldcroft, 2019-08-05:
I didn't realize (or forgot) the starcheck `pip install ..` in `build.py` is actually installing all the Perl source.   Anyway we can find a way to finesse this.

jeanconn, 2019-08-05:
I like that the conda install of starcheck does try to get the deps to give the user a working version of starcheck, but it wouldn't be in the end of the world if the install just installed the starcheck pieces and one would need to install ska3-perl to have the rest of the Perl packages necessary to run starcheck.  ska3-perl does have other pieces that aren't required to run starcheck though (and if we'd want to capture the dependencies someplace else we'd need something else to do that).

taldcroft, 2019-08-05:
Maybe remove `scripts=['starcheck/src/starcheck']` from setup.py and then install only that launcher script (and require the perl deps) as a separate package `starcheck-perl` or somesuch.

jeanconn, 2019-08-05:
So, your thought is two conda recipes to tease this apart along with two install paths to follow in the setup.py in the single starcheck repo?

I was leaning toward letting starcheck install without deps but update the `starcheck/src/starcheck` script so it has a useful error if installed without ska3-perl.

taldcroft, 2019-08-05:
> I was leaning toward letting starcheck install without deps but update the starcheck/src/starcheck script so it has a useful error if installed without ska3-perl.

That also sounds fine to me.

taldcroft, 2019-08-06:
For the record this issue does cause three proseco test fails in current ska3-flight without starcheck installed.

closed 2019-09-25 by jeanconn