sot/mica #216: Add guide star 5th, 16th, 50th, 84th, and 95th percentile mag columns to guide stats.

taldcroft, 2020-03-23:
Let's just do this now and start reprocessing the mission for https://github.com/sot/ska-projects/issues/114.

taldcroft, 2020-03-23:
BTW this processing is likely completely I/O bound and I would imagine it might run an order of magnitude faster on a local laptop with SSD.  Since it would be nice to have agile repro capability, might be worth the time to see if this can run on a laptop.

taldcroft, 2020-03-23:
With the idea the @javierggt could actually run it once @jeanconn makes it work (if it does not already).

jeanconn, 2020-03-23:
Great idea!

jeanconn, 2020-03-23:
Oh question though, this time around what do we want our mag convention to be for no data?  Every time I mask something I break all your speedy stuff.

taldcroft, 2020-03-23:
Call it 99.0 mag.  And yes, masked arrays can be more trouble than they're worth.

PR #219 (2020-04-14): Mag stats
## Description

Add 5, 16, 50, 84, 95 mag stats to the guide stats table.  This also adds some changes to support running on time ranges to make it easier to do trivial multiprocessing.

## Testing

- [x] Passes unit tests on MacOS, linux, Windows (at least one required)
- [x] Functional testing

Fixes #216
closed 2020-04-14 by jeanconn