sot/mica #229: shiny test warnings

jeanconn, 2020-08-07:
Not sure how to find and address this one

```
=============================== warnings summary ===============================
mica/archive/tests/test_asp_l1.py::test_get_atts_obsid[14333]
  <frozen importlib._bootstrap>:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject

-- Docs: https://docs.pytest.org/en/latest/warnings.html
=================== 6 passed, 1 warning in 221.12s (0:03:41) ===================
```

taldcroft, 2020-08-12:
You should not be seeing that in shiny due to:
https://github.com/sot/testr/blob/master/testr/runner.py#L15

Does your installed version have that exclusion in place?

jeanconn, 2020-08-12:
Well, it looks like my installed version was updated to testr-4.3.2.dev49+g53f4e6b-py38_0 8 minutes before this bug report, but I'm not seeing the issue now so closing ticket.

closed 2020-08-12 by jeanconn