sot/mica #292: Trouble finding starcheck catalog when commanded obsid not changing

jeanconn, 2024-02-08:
The logic here:

https://github.com/sot/mica/blob/45efdbdf24aa91cada860a6d498cadb0de7541bc/mica/starcheck/starcheck.py#L195

isn't compatible with "overlapping" kadi observations in the case where the obsid is not changing.  For example, get_starcheck_catalog_at_date isn't picking up the star catalog for 43729 as the obs_stop for 62619 is well after it.

(this is basically

ipdb> obss[46]
{'obsid': 62619, 'simpos': -99616, 'obs_stop': '2024:042:06:13:08.510', 'manvr_start': '2024:037:04:36:10.250', 'targ_att': (-0.14281354, -0.28517376, 0.84659287, 0.42609944), 'npnt_enab': True, 'obs_start': '2024:037:04:52:05.911', 'prev_att': (-0.0734461, -0.09617344, 0.91597702, 0.38254731), 'source': 'CMD_EVT'}
ipdb> obss[47]
{'obsid': 43729, 'simpos': -99616, 'obs_stop': '2024:038:01:49:18.954', 'manvr_start': '2024:038:01:15:10.251', 'targ_att': (-0.151278113, -0.301989774, 0.840740346, 0.423169917), 'npnt_enab': True, 'obs_start': '2024:038:01:20:47.178', 'prev_att': (-0.14281354, -0.28517376, 0.84659287, 0.42609944), 'starcat_date': '2024:038:01:15:06.000', 'starcat_idx': 229110, 'source': 'FEB0724A'}

Though maybe this is a just a kadi observations bug as the docs say "An “observation” is defined as a dwell in normal point mode " and the 62619 dwell is over and several science observations happen before its 'obs_stop' of '2024:042:06:13:08.510'.

javierggt, 2024-02-08:
I think this is a bug, and observations should never overlap.

jeanconn, 2024-02-08:
Right, I figure you can have multiple observations with the same obsid label if needed but an overlap is a bug, which I marked as https://github.com/sot/kadi/issues/318 without trying to figure out what the issue is in kadi yet.

jeanconn, 2024-09-26:
I think this was fixed in https://github.com/sot/kadi/pull/319 .

closed 2024-09-26 by jeanconn