sot/mica #313: Lots of missing `exp_time` values in local OCAT

taldcroft, 2024-12-19:
It looks like the exp_time values in the local OCAT file are frequently missing. Out of 50000 entries, about 18000 do not have `exp_time`. One obvious idea is that entries get ingested before the observation and then don't get updated. I haven't looked at the code yet. But I don't think there were 18000 observations since this feature was added.

jeanconn, 2024-12-29:
I looked at the update code and it just fetches everything from get_ocat_web and writes it out again.  It looks like this may just be the expected state for many observations (including ERs that end up in the ocat, and observations that are canceled or have not been observed).  It looks like there are 18534 observations with null exp_time but only 29 observations with null exp_time once I exclude ERs and observations with non-observed status (and the remaining 29 look to be CALs that may have odd configurations.
```
In [41]: dat = mica.archive.cda.get_ocat_web()

In [43]: np.count_nonzero(dat['exp_time'].mask)
Out[43]: 18534

In [42]: dat[(dat['exp_time'].mask) & (dat['type'] != 'ER') & (np.in1d(dat['status'], ['archived', 'observed']))]['obsid', 'status', 'target_name', 'type', 'a
    ...: pp_exp', 'exp_time']
Out[42]: 
<Table length=29>
obsid  status  target_name type app_exp exp_time
                                   ks      ks   
int64  str11      str39    str3 float64 float64 
----- -------- ----------- ---- ------- --------
   36 observed        VEGA  CAL     3.0       --
   37 observed        VEGA  CAL     3.0       --
 1855 observed  Betelgeuse  CAL     5.0       --
 1856 observed  Betelgeuse  CAL     5.0       --
 2586 observed        VEGA  CAL     3.0       --
 2587 observed        VEGA  CAL     3.0       --
 2593 observed  BETELGEUSE  CAL     5.0       --
 3366 observed  BETELGEUSE  CAL     5.0       --
 3367 observed  BETELGEUSE  CAL     5.0       --
 3678 observed  Betelgeuse  CAL     2.0       --
 3679 observed  Betelgeuse  CAL     2.0       --
 3685 observed        Vega  CAL     2.0       --
 3686 observed        Vega  CAL     2.0       --
 5046 observed        Vega  CAL     2.0       --
 5047 observed        Vega  CAL     2.0       --
 5048 observed  Betelgeuse  CAL     2.0       --
 5049 observed  Betelgeuse  CAL     2.0       --
 5961 observed        Vega  CAL     2.0       --
 5962 observed        Vega  CAL     2.0       --
 5963 observed  Betelgeuse  CAL     2.0       --
 5964 observed  Betelgeuse  CAL     2.0       --
 6445 observed        Vega  CAL     2.0       --
 6446 observed        Vega  CAL     2.0       --
 6447 observed  Betelgeuse  CAL     2.0       --
 6448 observed  Betelgeuse  CAL     2.0       --
17373 observed        Vega  CAL     2.0       --
17374 observed  Betelgeuse  CAL     2.0       --
17375 observed        Vega  CAL     2.0       --
17376 observed  Betelgeuse  CAL     2.0       --
```

taldcroft, 2025-01-06:
Thanks for looking into this. Sounds like everything is OK!